rauc/rauc-hawkbit-updater #17: Support for offline updates ?

gswebspace, 2020-01-13:
Hi,

There has been a discussion on offline deployments (ex: manual updates using a USB drive) leading to updating the UI regarding the update information, available [here](https://github.com/eclipse/hawkbit/pull/563) and [here](https://github.com/eclipse/hawkbit/issues/242).

Is this supported by `rauc-hawkbit-updater` when running as a daemon ?

prevas-lkmi, 2020-01-13:
No, it is not supported. But you are more than welcome to make a PR.

gswebspace, 2022-01-17:
Thanks for the update @prevas-lkmi and @Bastian-Krause . 

We decided to stick to online updates since that is what we want to do 99% of the times for devices in the field.

If anyone feels this functionality is critical to rauc-hawkbit-updater, they may open the issue again, but for now the issue can be closed.

closed 2022-01-17 by Bastian-Krause

rvdgracht, 2024-04-08:
Rauc has support for software redundancy. If booting a freshly installed distribution fails, it can fall back to the previous version. If this is not communicated back to Hawkbit, it will never know the update failed, and show the wrong version number.