rauc/rauc-hawkbit-updater #18: Moving Repository to 'rauc' GitHub Organization?

ejoerns, 2020-01-17:
Hi,

as this piece of software seems to raise more and more interest for people that look for a non-python alternative of https://github.com/rauc/rauc-hawkbit and do not want to reinvent the wheel, I would like to ask you if you are interested in moving the repository to the rauc organization on GitHub. This could give the project even more attention and makes it visibly belonging to RAUC.

We also plan to integrate this into meta-rauc and would like to clarify a future-proof location before ;)

Thus, if you are interested, we could move the repository keeping all issues etc intact and add you (@prevas-lkmi and other if desired) as maintainer for the project's team.

Would be great if you let me know which variant you prefer. If you have any concerns or questions, just let me know.

Best regards, Enrico

prevas-lkmi, 2020-01-20:
Hi Enrico

I see no problem in moving the repository to the RAUC organization, as long I get to be one of the maintainers. I would prefer to move the repository with all issues etc. if it is possible.

-Lasse

Raphexion, 2020-01-20:
I don't mind being a maintainer. You guys can decide.

ejoerns, 2020-01-20:
@prevas-lkmi Great to hear that, first of all and thanks for your reply!

Yes, transferring will preserve all the issues etc. And the idea is that you, and those of you that would like to can become or remain maintainers (and GitHub Collaborators) of the repo.

For actually transferring the repository it would be required to give me Administrator Collaborator rights on your repository. This can be done via 'Settings'->'Collaborators & Teams'. 
If we understood GitHub's permission handling correctly, then I should be allowed to transfer the repository.

Any concerns or questions regarding this?

prevas-lkmi, 2020-01-22:
> @prevas-lkmi Great to hear that, first of all and thanks for your reply!
>
> Yes, transferring will preserve all the issues etc. And the idea is that you, and those of you that would like to can become [...]

prevas-lkmi, 2020-01-22:
@ejoerns I have added you as admin now.

ejoerns, 2020-01-22:
:+1: let's move!

ejoerns, 2020-01-22:
And... welcome to rauc!

ejoerns, 2020-03-16:
I'd assume this can be closed now ;)

closed 2020-03-16 by ejoerns