rauc/rauc-hawkbit-updater #23: Segmentation fault caused by "config.conf" file

ctenruh-phytec, 2020-04-02:
Hi,
when using the rauc-hawkbit-updater, we get a SIGSEGV. Segmentation Fault.
We noticed, that the updater crashes while trying to get the data from the “config.conf” file.
It seems, that the Boolean variables in the “config.conf” file are not parsed correctly and because of that the comments, which are declared with a “#” on the same line, are being ignored as comments and written into the variables.

Did anyone have the same Problem?

Thanks.

ejoerns, 2020-04-02:
Comments (in the same line) are not (really) supported by the parser, yes.

Maybe we should remove them from the example, but it should work fine without them @ctenruh-phytec, isn't it?

In any way, it should not segfault for a broken config :)

ctenruh-phytec, 2020-04-02:
it works fine when the comments are removed. I will create a pull request for this issue with the edited config file.

PR #27 (2020-04-14): Fix Error Handling in config-file.c
This fixes the handling of GError's in config-file.c in many different ways.

As a result of this, no segmentation faults caused by unset errors should be triggered anymore and parsing of broken configuration should return valid error messages.

Fixes #23
Replaces #24
closed 2020-04-14 by prevas-lkmi