rauc/rauc-hawkbit-updater #33: potential SEGV with error

hpatriarche, 2020-05-11:
In function main, when config is NULL, it error could also be NULL so
                ```g_printerr("Loading config file failed: %s\n", error->message); ```
could crash

It miss a test error != NULL to  call error->message

also in load_config_file before returning NULL, it would be better to set error:
`g_set_error(error, xxx) `

like in each call like:
```        if (!g_key_file_load_from_file(ini_file, config_file, G_KEY_FILE_NONE, error))
                return NULL;
```

ejoerns, 2020-05-11:
@hpatriarche Where can `config` be `NULL` while `error` isn't?

> also in load_config_file before returning NULL, it would be better to set error:
g_set_error(error, xxx)

Could you point to a line where this is not the case?

This is the default way of GError handling for glib. A function that returns NULL to signalize an error must set GError. Otherwise you would have to add these kind of checks everywhere in the code.

hpatriarche, 2020-05-11:
my bad, it was on my changes.

closed 2020-05-11 by hpatriarche