rauc/rauc-hawkbit-updater #40: Curl Follow Location

hpatriarche, 2020-05-14:
Hi,

In our version, we added 
curl_easy_setopt(curl, CURLOPT_FOLLOWLOCATION, 1L);

is it an option that make sense by default ?

jluebbe, 2020-05-14:
We have that in rauc's network code as well. So I'd say yes...

Raphexion, 2020-05-14:
I think it makes a lot of sense. Is there any risk adding this?

ejoerns, 2020-07-02:
Many projects set this [1], thus I don't see why we shouldn't.
I'd expect location following to be the common default and not following it only for specific reasons.
The reasons it is not the default in libcurl to my knowledge is only that the libcurl default way is to have almost everything turned off unless explicitly enabled.
As documentation and examples state, this should be used together with setting `CURLOPT_MAXREDIRS` to a low number (most use `5` here).

[1] https://codesearch.debian.net/search?q=CURLOPT_FOLLOWLOCATION&literal=1

PR #56 (2020-09-08): src/hawkbit-client: allow following HTTP redirects
There is no special reason for not doing this and most tools using curl
have this enabled (including rauc) thus we allow redirects here too,
but with a limit of 5 redirects which seems to be a common default.

Fixes #40
commit 904a4508 (2020-09-08): src/hawkbit-client: allow following HTTP redirects

There is no special reason for not doing this and most tools using curl
have this enabled (including rauc) thus we allow redirects here too,
but with a limit of 5 redirects which seems to be a common default.

Fixes #40
closed 2020-09-08 by prevas-lkmi