rauc/rauc-hawkbit-updater #43: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

hpatriarche, 2020-06-26:
This error is coming from;
```
static gboolean process_deployment(JsonNode *req_root, GError **error)
```
in case  json_response_parser is NULL
```
        if (status != 200 || json_response_parser == NULL) {
                g_debug("Failed to get resource from hawkbit server. Status: %d", status);
                goto proc_error;
        }
```
jumping in proc_error:
```
proc_error:
        g_object_unref(json_response_parser);
```

mschwan-phytec, 2020-06-30:
One cannot pass `NULL` to `g_object_unref()`. Does this fix the issue for you?
```diff
diff --git a/src/hawkbit-client.c b/src/hawkbit-client.c
index 6802ac2..72589c5 100644
--- a/src/hawkbit-client.c
+++ b/src/hawkbit-client.c
@@ -704,7 +704,9 @@ static gboolean process_deployment(JsonNode *req_root, GError **error)
         return TRUE;

 proc_error:
-        g_object_unref(json_response_parser);
+        if (json_response_parser != NULL) {
+                g_object_unref(json_response_parser);
+        }
         // Lets cleanup processing deployment failed
         process_artifact_cleanup(artifact);
         process_deployment_cleanup();
```

hpatriarche, 2020-07-01:
I have same fix on my sandbox.

ejoerns, 2020-07-01:
This part of the code needs a major rework to my opinion, but having this distinct bug fixed first should not prevent us from that.

However, this is also fixable by not going to `proc_error` if `rest_request()` fails butdirectly returning `FALSE` as in the error paths above.

The implementation of rest_requests() ensures that it only sets `json_response_parser` if `status == 200`. Thus there is nothing more or less we need to free than in the error cases  above.

ejoerns, 2020-09-04:
Du to the used syntax, this was not automatically closed (but fixed) by #51.

We can close this manually now.

closed 2020-09-04 by ejoerns