rauc/rauc-hawkbit-updater #44: systemd service assuming user and group "rauc-hawkbit"

mschwan-phytec, 2020-06-30:
Is there a specific reason for specifying a user and group for the systemd service?
```
[Service]
User=rauc-hawkbit
Group=rauc-hawkbit
```
This assumes the target system has such a user, which I'd argue is not the case for almost all systems by default. Would it be an option to leave this out and let users add their own user and group if they want? Otherwise the service simply will not start.

@prevas-lkmi Can you elaborate on this?

ejoerns, 2020-07-02:
I guess this should prevent from running rauc-hawkbit as root, which makes sense in terms of system hardening. Systemd does this too, by default, e.g. for systemd-timesyncd.

I haven't tested yet what this would mean for the user configuration. Just adding the user is something one could simply do in a Linux build system recipe, but we should make sure this actually works somehow then.

mschwan-phytec, 2020-07-02:
Simply issuing `useradd rauc-hawkbit` makes the service run, so I guess there is nothing special to it.

I can prepare a pull-request in meta-rauc, if that's appropriate?

closed 2020-07-08 by ejoerns