rauc/rauc-hawkbit-updater #45: hawkbit-client.c: memory leak on debug messages

rfuentess, 2020-08-14:
hawkbit-client. c is making use of `json_to_string()` for debugging messages ([1](https://github.com/rauc/rauc-hawkbit-updater/blob/master/src/hawkbit-client.c#L641), [2](https://github.com/rauc/rauc-hawkbit-updater/blob/master/src/hawkbit-client.c#L750)). However, this is provoking a small leak as said function returns a `char *` that must be released manually. 

A possible fix could be:

```
char *str = json_to_string(resp_root, TRUE);
g_debug("Deployment response: %s\n", str);
free(str);
```

If this approach is OK, we can proceed to generate a PR :)

ejoerns, 2020-08-14:
@rfuentess You're right, this is still an issue.

I'd suggest making use of autofree as we already do in other locations, too. Thus something like

``` c
g_autofree gchar *str = json_to_string(resp_root, TRUE);
g_debug("Deployment response: %s\n", str);
```

rfuentess, 2020-08-17:
I'm new to this macro.  But because `gchar` is an alias for `char`  I think we are going to be fine :)
I'll run some tests just to be sure and then I'll proceed to generate the PR

PR #47 (2020-08-18): hawkbit-client: Fix memory leak on debug messages
This fixes #45 (memory leak on debug messages). 

NOTE: Although I detected the leak, it was @hpatriarche  who truly was able to trace it to its source. Therefore, I'm just the committer.
closed 2020-08-18 by prevas-lkmi