rauc/rauc-hawkbit-updater #60: Stable version

einoj, 2020-11-06:
Could you make tags for stable versions of this software? [meta-rauc](https://github.com/rauc/meta-rauc/blob/dunfell/recipes-support/rauc-hawkbit-updater/rauc-hawkbit-updater_git.bb) is currently supplying an unstable version of rauc-hawkbit-updater, commit d909982e9e which does not include https://github.com/rauc/rauc-hawkbit-updater/pull/35. It would save time debugging if I realized that they were shipping an old unstable version, this could have been prevented if this project used version tags, at leas tagged commits that are considered stable.

Bastian-Krause, 2021-01-27:
We're currently in the process of refactoring the code base, transforming the tests to pytest (which allows us actual download and installation tests) and adding documentation. Once we're done with this, we can prepare a release. I'd like to be able to create a release tarball, so we need to integrate support for that first.

ejoerns, 2021-06-04:
We are now at a point where cleanup and rework are done from my perspective. We've addressed the above-mentioned show stoppers for a release (powerful pytest-based tests and basic documentation). Maybe that's the point now for creating a release and adding new features afterwards.

@prevas-lkmi what do you think about that?

Bastian-Krause, 2021-06-04:
Do we need mechanics to create a release tarball? And what about #82?

If the solutions to those questions require more effort than we are currently able to carry out, I'm in favor of tagging a release "as is" rather than waiting for another couple of months.

prevas-lkmi, 2021-06-07:
I agree it would be nice to create a "as-is" release now. But before doing that, we should consider to update VERSION in rauc-hawkbit-updater.c? 1.0?

ejoerns, 2021-06-07:
Yes, something above 0.1 (which we have in all versions up to now already) sounds appropriate. From my perspective 1.0 sounds quite mature and despite some features are still missing I would consider the code as 'quite mature' now. So fine for 1.0 from my side (before ending up as a project that's never brave enough to switch :D ).

Additional things to care about:

* adapt CHANGES file to 1.0
* adapt 'release' in conf.py
* have a look at https://github.com/rauc/rauc/blob/master/docs/release-checklist.txt probably, e.g. for creating signed releases (if desired)

I don't have any experience with packaging a CMake release. The only option I would know is `git archive`. Are there more advanced solutions, like a 'make dist' from autotools?

Anything else we forgot? @Bastian-Krause ?

Bastian-Krause, 2021-06-07:
It would be cool to have #90 included. Apart from that, no, can't think of anything we forgot.

ejoerns, 2021-06-08:
Fine!

@prevas-lkmi do you take control over the finalization process for the release? Or do you want me/us to care about this?

prevas-lkmi, 2021-06-08:
@ejoerns I can do the finalization process. No problem.

Bastian-Krause, 2021-06-13:
@prevas-lkmi I just stumbled upon a use-after-free and a locking issue. Could you please wait with the release wait until these bugs are squashed? Thanks!

prevas-lkmi, 2021-06-14:
@Bastian-Krause, no problem.

Bastian-Krause, 2021-06-17:
@prevas-lkmi Could you have a look at #92 and #93? If these PRs are merged, we should be good to go.

ejoerns, 2021-09-15:
We just released v1.0. So this can be closed now :tada: 

Thanks to all contributors and especially to @prevas-lkmi for founding this project!

closed 2021-09-15 by ejoerns