rauc/rauc-hawkbit-updater #62: Any suggestion support install casync bundle from hawkbit?

sprhawk, 2020-11-28:
I'm building casync bundle for minimizing updates size.

But I see hawkbit only support upload files not directory of `castr`

So can I specify location of `castr` or other means to do update casync bundle via  hawkbit?

ejoerns, 2020-11-30:
@sprhawk Yes, the way this is meant to work is that hawkBit provides you only the RAUC bundle containing the chunk index file. Access to your server's chunk store happens fully independent of hawkbit.

When not using hawkBit, the default lookup location would be right next to the original bundle URL on the server. This information about the original bundle location is actually lost when using hawkBit, as it puts the bundle in a temporary download location first and then notifies RAUC. However, what is possible but is not part of RAUC (yet) is that hawkbit tells RAUC where to expect the store path during the 'Install' D-Bus call. There is a pending PR for this: https://github.com/rauc/rauc/pull/630

Another option is what you already discovered, having a fixed store path configured in the system configuration. With migrating between different versions of castr you refer to possible store format changes?
When you only refer to having multiple bundle versions, then this is not an issue in most cases as you can simply put all chunks for all versions of a bundle into a single chunk store. This is another major advantage of casync: it allows you to save storage space on the server side, too. Thus if that is possible in your setup, this is probably the way to go.

The last option I could think of is having a store path set in the bundle's manifest file. This would at least allow to pre-set it during bundle generation but would require some pre-knowledge of server setup possibly not yet available a the time of bundle generation.

Enrico

sprhawk, 2020-12-04:
@ejoerns thanks.

closed 2020-12-04 by sprhawk