rauc/rauc-hawkbit-updater #78: GError HTTP Request Failed with Server

MitchGaines, 2021-03-24:
Hi there,

I have been struggling with the `rauc-hawkbit-updater` service for a few days now and keeping hitting the error below when I try to deploy a new artifact.

Here is my rauc-config.conf file:
```
root@raspberrypi4:~# cat rauc-config.conf
[client]
# host or IP and optional port
hawkbit_server = ***.com
ssl                       = false
ssl_verify                = false
tenant_id                 = DEFAULT
target_name               = dev01 
auth_token                = abc123 #dummy value for actual token 
bundle_download_location  = /tmp/bundle.raucb
retry_wait                = 60
connect_timeout           = 20
timeout                   = 60
log_level                 = debug 
post_update_reboot        = false 

[device]
mac_address               = MACADDRESS
hw_revision               = 1
model                     = dev01
```

Whenever I try to run with the above config file, this is the result:
```
root@raspberrypi4:~# rauc-hawkbit-updater -d -c rauc-config.conf
MESSAGE: Checking for new software...
DEBUG: Response body: {
  "config" : {
    "polling" : {
      "sleep" : "00:05:00"
    }
  },
  "_links" : {
    "deploymentBase" : {
      "href" : "http://***.com/DEFAULT/controller/v1/dev01/deploymentBase/2?c=412523400"
    },
    "configData" : {
      "href" : "http://***.com/DEFAULT/controller/v1/dev01/configData"
    }
  }
}
DEBUG: Identifying ourself to hawkbit server
DEBUG: Request body: {
  "time" : "20210324T125933",
  "status" : {
    "result" : {
      "finished" : "success"
    },
    "execution" : "closed"
  },
  "data" : {
    "model" : "dev01",
    "mac_address" : "MACADDRESS",
    "hw_revision" : "1"
  }
}

(rauc-hawkbit-updater:421): GLib-WARNING **: 12:59:33.452: GError set over the top of a previous GError or uninitialized memory.
This indicates a bug in someone's code. You must ensure an error is NULL before it's set.
The overwriting error message was: HTTP request failed: No error
DEBUG: Failed to get resource from hawkbit server. Status: 400
CRITICAL: Error: HTTP request failed: No error
```

Based on that response, it seems to be successfully communicating with my server. As for why it is failing to pull the new artifact I am deploying, I cannot figure out. Am I doing something blatantly wrong here? The error - `CRITICAL: ERROR: HTTP request failed: No error` - is quite... frustrating to debug with :sweat_smile: 

Thank you for you time and assistance

Bastian-Krause, 2021-03-24:
Hi there!

What version of rauc-hawkbit-updater are you using?

This coding issue should be solved with PR #77. Could you test this?

Bastian-Krause, 2021-03-24:
Trying a shot in the dark: the `target_name` line in your config contains a trailing space, try to remove it. I guess the comment in the `auth_token` line is not there in your actual config, right? Make sure, there is no trailing space in that line either (and in all other lines).

MitchGaines, 2021-03-24:
@Bastian-Krause Thanks for the quick response on this.

Looks like I'm running version 0.1:

```
root@raspberrypi4:~# rauc-hawkbit-updater -v
Version 0.1
```

Thanks for your shot in the dark here. Looks like the trailing space was an issue :facepalm: 

However,  when I removed the trailing space it has presented another issue. So, at the risk of being one of those "ask a million unrelated questions on a single issue" type of people, I'm just going to fire away here.. maybe you can assist:

```
root@raspberrypi4:~# rauc-hawkbit-updater -d -c rauc-config.conf
MESSAGE: Checking for new software...
DEBUG: Response body: {
  "config" : {
    "polling" : {
      "sleep" : "00:05:00"
    }
  },
  "_links" : {
    "deploymentBase" : {
      "href" : "http://***.com/DEFAULT/controller/v1/dev01/deploymentBase/8?c=418064526"
    }
  }
}
DEBUG: Response body: {
  "id" : "8",
  "deployment" : {
    "download" : "forced",
    "update" : "forced",
    "chunks" : [
      {
        "part" : "os",
        "version" : "1.1",
        "name" : "***",
        "artifacts" : [
          {
            "filename" : "update-bundle-raspberrypi4.raucb",
            "hashes" : {
              "sha1" : "***",
              "md5" : "***",
              "sha256" : "***"
            },
            "size" : 133793178,
            "_links" : {
              "download-http" : {
                "href" : "http://***.com:8080/DEFAULT/controller/v1/dev01/softwaremodules/3/artifacts/update-bundle-raspberrypi4.raucb"
              },
              "md5sum-http" : {
                "href" : "http://***.com:8080/DEFAULT/controller/v1/dev01/softwaremodules/3/artifacts/update-bundle-raspberrypi4.raucb.MD5SUM"
              }
            }
          }
        ]
      }
    ]
  }
}
MESSAGE: New software ready for download. (Name: ***, Version: 1.1, Size: 133793178, URL: http://***.com:8080/DEFAULT/controller/v1/dev01/softwaremodules/3/artifacts/update-bundle-raspberrypi4.raucb)
MESSAGE: Start downloading: http://***.com:8080/DEFAULT/controller/v1/dev01/softwaremodules/3/artifacts/update-bundle-raspberrypi4.raucb
CRITICAL: Download failed: HTTP request failed: Timeout was reached Status: 0
DEBUG: Request body: {
  "id" : "8",
  "time" : "20210324T153908",
  "status" : {
    "result" : {
      "finished" : "failure"
    },
    "execution" : "closed",
    "details" : [
      "Download failed: HTTP request failed: Timeout was reached Status: 0"
    ]
  }
}
DEBUG: Feedback status: 200, URL: http://***.com/DEFAULT/controller/v1/dev01/deploymentBase/8/feedback
```

I have tried making all the timeouts in the rauc-config.conf really high (I set them all to 600). It doesn't seem to really impact how long it takes to time out. Is another config parameter that would fix this?

It seems to be establishing an okay connection. Worth mentioning that the artifact I'm pulling is 134 MB.

ejoerns, 2021-03-24:
@MitchGaines Currently, all revisions identify themself as 0.1 unfortunately. Do you know from which git / BSP / recipe version you built that? Please try to update to current master if it isn't already and check if things change.

From what I see in the log I would assume that the download attempt failed completely. Maybe check if manually downloading via curl or with a much smaller artifact works.

I'm also a bit confused that the actionID suddenly changes from '3' to '8' in your logs. Did you cut out a bunch of log lines?

Bastian-Krause, 2021-03-25:
> Download failed: HTTP request failed: Timeout was reached Status: 0

You're running into `CURLE_OPERATION_TIMEDOUT` (see [here](https://github.com/curl/curl/blob/8593b15c77f7f2a73512261a0b0b97a64e5e93c8/lib/strerror.c#L144)). [Curl's documentation](https://curl.se/libcurl/c/libcurl-errors.html) reads:

> Operation timeout. The specified time-out period was reached according to the conditions.

That means your download is slower than 100 bytes/sec during 60 seconds (see [here](https://github.com/rauc/rauc-hawkbit-updater/blob/5a55ab7bc5ac1f14d8938fb3475dc13a181fab06/src/hawkbit-client.c#L155)). This is currently not configurable. I'll make this configurable in a future PR though.

So please allow downloading of artifacts without authentication temporarily (hawkBit UI -> System Config -> [x] Allow targets to download artifacts without security credential) and use wget/curl as suggested by @ejoerns to reproduce that.

MitchGaines, 2021-03-25:
> @MitchGaines Currently, all revisions identify themself as 0.1 unfortunately. Do you know from which git / BSP / recipe version you built that? Please try to update to current master if it isn't already and check if things change.

Built from `dunfell` of `meta-rauc`. To update to current master, I have modified `meta-rauc/recipes-support/rauc-hawkbit-updater/rauc-hawkbit-updater_git.bb:7` to specify the master branch of this repo as follows (though, I _think_ this is the default already - I'm not 100% sure though): 
`SRC_URI = "git://github.com/rauc/rauc-hawkbit-updater.git;protocol=https;branch=master"` 

After doing this, I appear to still be hitting the same issue.

> That means your download is slower than 100 bytes/sec during 60 seconds (see [here](https://github.com/rauc/rauc-hawkbit-updater/blob/5a55ab7bc5ac1f14d8938fb3475dc13a181fab06/src/hawkbit-client.c#L155)).

As far as I can tell, this will be the issue. Will have to play around with this and see what I can make happen.

> This is currently not configurable. I'll make this configurable in a future PR though.

Thank you!

Will go ahead and close this as it seems the questions have been answered. Thank you, @Bastian-Krause and @ejoerns for your time and assistance in this. I really appreciate it :smile:

closed 2021-03-25 by MitchGaines

ejoerns, 2021-03-25:
> Built from `dunfell` of `meta-rauc`. To update to current master, I have modified `meta-rauc/recipes-support/rauc-hawkbit-updater/rauc-hawkbit-updater_git.bb:7` to specify the master branch of this [...]

The relevant line here is the `SRCREV` setting!

https://github.com/rauc/meta-rauc/blob/dunfell/recipes-support/rauc-hawkbit-updater/rauc-hawkbit-updater_git.bb#L8

The state used there is 28900b59859d80d2e5275ad03278b0b76e7f1b51 while the recent master is 5a55ab7bc5ac1f14d8938fb3475dc13a181fab06.

MitchGaines, 2021-03-25:
Ahh yes, gotcha. Okay, so I have changed it to be the most recent master. Unfortunately, still experience the same behavior. 

Going to need to play around getting curl to download the artifact. It also seems to be timing out even with the `--connect-timeout 600` added onto it. Odd.

```
root@raspberrypi4:/tmp# curl -i -X GET http://***:8080/DEFAULT/controller/v1/dev01/softwaremodules/4/artifacts/update-bundle-raspberrypi4.raucb -H 'Authorization: TargetToken ***' --connect-timeout 600
curl: (28) Failed to connect to ***.com port 8080: Connection timed out
```