rauc/rauc-hawkbit-updater #85: updater use http for deploymentBase link instead of https (301 Moved Permanently)

motto-phytec, 2021-04-22:
rauc-hawkbit-updater can connect successfull with the hawkbit. check for new software is ok.
the rauc-hawkbit-updater do not use https for the deploymentBase. My old version (commit id (28900b59859d80d2e5275ad03278b0b76e7f1b51) use https and it runs.
rauc-hawkbit-updater -d -c /mnt/config/hawkbit/config.cfg
MESSAGE: Checking for new software...
DEBUG: Response body: {
"config" : {
"polling" :
{ "sleep" : "00:05:00" }

},
"_links" : {
"deploymentBase" :
{ "href" : "http://hawkbit.xxxx/DEFAULT/controller/v1/xxxxxx/deploymentBase/1?c=411599879" }

,
"configData" :
{ "href" : "http://hawkbit.xxxxx/DEFAULT/controller/v1/xxxxx/configData" }

}
}
DEBUG: Providing meta information to hawkbit server
DEBUG: Request body: {
"time" : "20210420T142812",
"status" : {
"result" :
{ "finished" : "success" }

,
"execution" : "closed"
},
"data" :
{ "mac_address" : "00:00:00:00:00:00" }

}
WARNING: HTTP request failed: 301; server response: <html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
</body>
</html>

config file is
[client]
hawkbit_server = hawkbit.xxxx
ssl=true
ssl_verify = true
tenant_id = DEFAULT
target_name = xxxx
auth_token = xxxx
bundle_download_location = /tmp/hawkbit.raucb
log_level = message

[device]
mac_address = 00:00:00:00:00:00

Bastian-Krause, 2021-04-22:
Hi,

the `depoymentBase` URL is used as given by hawkBit since 8624e71a. Before that the URL was handcrafted. In my opinion the correct way is to use the URLs as provided by hawkBit.

I guess you are using some kind of SSL reverse proxy that hawkBit is not aware of?

motto-phytec, 2021-04-23:
Hi

thank you for your help.  A configuration in hawkbit with the credentials seems the reason for the problem. 
so the issue can be closed

closed 2021-04-23 by motto-phytec

Bastian-Krause, 2021-04-23:
Thanks for the feedback. Could you tell us which hawkBit option is the solution for your issue?

motto-phytec, 2021-04-27:
The option in the “application.properties" 
`server.use-forward-headers=true`
will ignored with the migration to Springboot 2.2 at hawkbit 0.3.0M7 
The replacement is
`server.forward-headers-strategy=NATIVE`

Bastian-Krause, 2021-04-27:
Thanks for pointing out the solution!