rauc/rauc-hawkbit-updater #87: HTTP Request failed with 301

hmenn, 2021-05-04:
Hi, I've decided to use newer version of updater in our system but I got an error.

I was using b38f5a580ab66845d7376b276c80da4b2396337d previously and Now, using latest commit cc2cd6e2f55c547d78a365f024a2ab5e40bbb22e.

I've also tested 5a55ab7bc5ac1f14d8938fb3475dc13a181fab06 but no success.

Here's the log message that I've got

```
root@DEV:~# /usr/bin/rauc-hawkbit-updater -d -c /userdata/rauc-hawkbit-updater/config.conf
MESSAGE: Checking for new software...
DEBUG: Response body: {
  "config" : {
    "polling" : {
      "sleep" : "00:05:00"
    }
  },
  "_links" : {
    "deploymentBase" : {
      "href" : "http://URL/default/controller/v1/MY_DEVICE_ID/deploymentBase/918?c=1258468636"
    }
  }
}
WARNING: HTTP request failed: 301; server response: <html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>nginx/1.17.5</center>
</body>
</html>

```

Does anyone has an idea?

hmenn, 2021-05-04:
A small update: a5d0bda1aa37107fa59937f363d231ae6dc3726e work but next commit 91e9701959b132f3825df2289f2ec1fea94553a7 doesn't work. It looks like my problem is related with #68

hmenn, 2021-05-04:
It looks like problem is related with Hawkbit Server but We've realized that after changes on process_deployment and build_api_url. Previous implementation was using some part of ```$._links.deploymentBase.href``` and config file defitions to produce new deployment URL but currently it uses what server send in response. Strange thing is my server returns http path instead of https address.

I decided to continue with a6fe7acdd66c5ed11e540578a36ac3401d603140 until I found a fix or upgrade server.

Bastian-Krause, 2021-05-05:
A similar issue was reported in #85. Could you try the solution mentioned there?

Bastian-Krause, 2021-06-02:
@hmenn Did you try the solution I mentioned?

hmenn, 2021-06-03:
@Bastian-Krause I've opened an upgrade request to our Hawkbit Server Maintainence team. I'm still waiting response from them until them I think we can close issue. I'm using old version of rauc-hawkbit-updater for now.

Bastian-Krause, 2021-06-03:
Okay, feel free to reopen if the issue persists.

closed 2021-06-03 by Bastian-Krause