rauc/rauc-hawkbit-updater #107: Feature Wish List

ejoerns, 2021-09-15:
Note: This was in the Wiki before as 'Roadmap'. Moved it to an issue to close the (unused) wiki but not lose these items.

- [ ] "Ask" permission to download (D-BUS).
- [ ] "Ask" permission to install (D-BUS).
- [ ] Expose download and install progress via D-BUS (like Rauc service does).
- [ ] Support casync

jluebbe, 2021-09-15:
> Note: This was in the Wiki before as 'Roadmap'. Moved it to an issue to close the (unused) wiki but not lose these items.
> 
> * [ ]  "Ask" permission to download (D-BUS).
> * [ ]  "Ask" permission to install (D-BUS).

Why should this be in rauc-hawkbit-updater instead of rauc itself?

> * [ ]  Expose download and install progress via D-BUS (like Rauc service does).
> * [ ]  Support casync

Isn't this better handled by using RAUC's streaming mode? I wouldn't like to just wrap the existing rauc D-Bus API.

ejoerns, 2021-09-15:
> > Note: This was in the Wiki before as 'Roadmap'. Moved it to an issue to close the (unused) wiki but not lose these items.
> > 
> > * [ ]   "Ask" permission to download (D-BUS).
> > * [ ]   "Ask" permission to install (D-BUS).
> 
> Why should this be in rauc-hawkbit-updater instead of rauc itself?

The real download of a bundle (assuming not using streaming support) cannot be handled by RAUC as it happens before RAUC gets actually called. Asking permission for installation instead could be handled by RAUC.
A possible drawback could be that Hawkbit would assume the installation is started and in progress as long as the update was not confirmed (assuming physical interaction e.g.). Maybe this is acceptable/intended.

> > * [ ]   Expose download and install progress via D-BUS (like Rauc service does).
> > * [ ]   Support casync
> 
> Isn't this better handled by using RAUC's streaming mode? I wouldn't like to just wrap the existing rauc D-Bus API.

When being merged, RAUC streaming mode is possibly preferred, yes (despite it still lacks some additional features). But that was not available at the time of writing this list.
The comment about wrapping the D-Bus API targets the first item? Or how does this deal with casync?

closed 2021-09-15 by ejoerns