rauc/rauc-hawkbit-updater #109: downloaded bundle has the wrong owner

j-licht, 2021-09-16:
The downloaded bund is saved with the owner root:rauc-hawkbit-updater and permissions 644.
rauc-hawkbit-updater is running by the systemd service file.
So the updater fails to delete the old bundle either to download a new one.

j-licht, 2021-09-17:
I think to owner is changing in the update process by the rauc tool running as root.
So what's the correct way to run rauc-hawkbit-updater as unprivileged user?

jluebbe, 2021-09-20:
If the download directory belongs to the rauc-hawkbit-updater user, it should be able to delete the bundle.

j-licht, 2021-09-20:
The example config uses /tmp/bundle.raucb.
So should be the right way to use a subfolder of /tmp?

jluebbe, 2021-09-20:
The proper location depends on your expected bundle sizes and where you have enough temporary space for the download.

j-licht, 2021-09-20:
OK. I have a readonly rootfs and the /tmp dir is a tmpfs.
As I see rauc-hawkbit-updater doesn't create the tmp dir, so should I add a mkdir to the service file?

Bastian-Krause, 2021-09-23:
Seems like the correct approach for your use case.

Bastian-Krause, 2021-10-04:
@j-licht Did that work out for you? Is there anything that would have made figuring it out easier (like a comment in the docs)?

j-licht, 2021-10-05:
I'm not quite sure in which scenario the default config with download location /tmp/bundlle.rauc and default service file running rauc-hawkbit-updater as unprivileged user is working at all.
My suggestion is to change the default config and service file to use a subfolder of /tmp, or I missed something?
I can open a PR if this helps.

Bastian-Krause, 2021-10-05:
RAUC only takes ownership of [plain format bundles](https://rauc.readthedocs.io/en/latest/reference.html#plain-format). If you use [verity format bundles](https://rauc.readthedocs.io/en/latest/reference.html#verity-format) (the [recommended](https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration) format), the example download location works as is.

Since we are talking about the example location here, I'd like to keep this separate from the systemd service file. I could imagine an exemplary suggestion for a systemd drop-in to cover plain bundle cases though.

Bastian-Krause, 2022-04-04:
@j-licht Right, would you care to create a PR to change the bit in the documentation?