rauc/rauc-hawkbit-updater #117: Temporarily no network connectivity: Scheduled check for new software failed

lab-w, 2022-01-06:
Hello,
thanks for this nice piece of software. I just switched from rauc-hawkbit-client to rauc-hawkbit-updater and noticed a bug:
when I execute rauc-hawkbit-updater from an autostart script it only prints these two lines and after that it does not retry to connect again:
```
MESSAGE: Checking for new software...
DEBUG: Scheduled check for new software failed status code: 0
```
probably this line is also printed (but not written to my log file):
```
CRITICAL: HTTP Error: HTTP request failed: Couldn't resolve host name
```

It is possible that at that point in time the network interfaces of the system are not up or configured properly, but with rauc-hawkbit-client that was not a problem - a later retry did work there. I encounter the same problem when I shut down the ethernet interface and then start the application manually.

When I then kill rauc-hawkbit-updater and execute it again manually (when ethernet is active) it does connect to the server. Any ideas how to fix this?
Thanks!

lab-w, 2022-01-06:
I just noticed that there is in fact a retry after some minutes so the connectivity is there when I wait for some more minutes than I expected.

closed 2022-01-06 by lab-w

ejoerns, 2022-01-06:
@lab-w Glad it works for you now.

Please consider using a more recent version of rauc-hawkbit-updater. The error message means that you use a version from before the initial release of this project.

We've added a lot of options and enhancements concerning retry/resume handling since this and fixed a number of issues.

Bastian-Krause, 2022-01-06:
Also note that you can decrease the time rauc-hawkbit-updater waits in these cases with the `retry_wait` config option, see [here](https://rauc-hawkbit-updater.readthedocs.io/en/latest/reference.html#configuration-file). The default for that value seems to documented incorrectly, though. It is 300 s, not 60 s.

lab-w, 2022-01-06:
Thank you both for the quick replys.

@ejoerns Thank you for the hint - updating the client is on my list ;-)

@Bastian-Krause Thanks! This solves my "problem". Now I reduced the timeout in the config file and it works like it should.