rauc/rauc-hawkbit-updater #121: hawkbit URL regression

einoj, 2022-01-31:
I am unable use the new tags v1.0 and v1.1. I've identified the issue stemming from a regression caused by commit 8624e71accf751c370ffe4a3f12ee220a103be79 

I had a look at the old code https://github.com/rauc/rauc-hawkbit-updater/blob/cbafe6522b671b0f34a86779e98b48d92ca40b4f/src/hawkbit-client.c#L638
and the new https://github.com/rauc/rauc-hawkbit-updater/blob/8624e71accf751c370ffe4a3f12ee220a103be79/src/hawkbit-client.c#L605
and printed the variables `deployment`  in the new version and `get_resource_url` in the old version, and see that in the new `deployment` variable the domain name and TLD of the URL is missing.

```
(gdb) print deployment
$1 = (gchar *) 0xaaaaaabc7100 "http://hawkbit:8080/DEFAULT/controller/v1/TEST/deploymentBase/2065?c=-1956918325"  `
```

```
(gdb) print get_resource_url
$1 = (gchar *) 0xaaaaaaceb560 "https://hawkbit.mydomain.se/DEFAULT/controller/v1/TEST/deploymentBase/2065?c=-1956918325"
```

Bastian-Krause, 2022-01-31:
Hi,

please run rauc-hawkbit-updater with `--debug`. This should print the json responses from the server. I'd like to see the response to the base resource, similar to [this example](https://www.eclipse.org/hawkbit/rest-api/rootcontroller-api-guide/#_response_example).

Bastian-Krause, 2022-01-31:
Might be similar to #85.

einoj, 2022-02-01:
Hi, 

Thanks for the feedback. With the --debug flag running v1.1 I get the same result as with GDB

```
MESSAGE: Checking for new software...
DEBUG: Response body: {
  "config" : {
    "polling" : {
      "sleep" : "23:59:00"
    }
  },
  "_links" : {
    "deploymentBase" : {
      "href" : "http://hawkbit:8080/DEFAULT/controller/v1/TEST/deploymentBase/2137?c=-1956918325"
    }
  }
}
WARNING: Couldn't resolve host name
```

I'll have a look at the example and #85

einoj, 2022-02-01:
with commit https://github.com/rauc/rauc-hawkbit-updater/commit/8624e71accf751c370ffe4a3f12ee220a103be79 the error message is slightly different

```
MESSAGE: Checking for new software...
DEBUG: Response body: {
  "config" : {
    "polling" : {
      "sleep" : "23:59:00"
    }
  },
  "_links" : {
    "deploymentBase" : {
      "href" : "http://hawkbit:8080/DEFAULT/controller/v1/TEST/deploymentBase/2137?c=-1956918325"
    }
  }
}
DEBUG: Failed to get resource from hawkbit server. Status: 0
CRITICAL: Error: HTTP request failed: Couldn't resolve host name
```

And with the parent commit https://github.com/rauc/rauc-hawkbit-updater/commit/cbafe6522b671b0f34a86779e98b48d92ca40b4f it works fine

```
MESSAGE: Checking for new software...
DEBUG: Response body: {
  "config" : {
    "polling" : {
      "sleep" : "23:59:00"
    }
  },
  "_links" : {
    "deploymentBase" : {
      "href" : "http://hawkbit:8080/DEFAULT/controller/v1/TEST/deploymentBase/2137?c=-1956918325"               
    }
  }                                                           
}                                                             
DEBUG: Response body: {                                       
  "id" : "2137",                     
  "deployment" : {        
    "download" : "forced",
    "update" : "forced",
    "chunks" : [             
      {       
        "part" : "os",
        "version" : "v1.2.3",
        "name" : "test-linux",
        "artifacts" : [  
          {      
            "filename" : "test-linux-delta.raucb",
            "hashes" : {
              "sha1" : "dcf25521b1ca54da339354bb98d25daab51d767e",                            
              "md5" : **"a8ce7ac760017c4bd7e3173f0ce6a6f30",**     
              "sha256" : "6bbf1a9ee6025a44974561cc63366e7b950503f71e44d694383f21ef9539163d"
            },                 
            "size" : 143726,              
            "_links" : {                           
              "download-http" : {
                "href" : "https://hawkbit.mydomain.se/DEFAULT/controller/v1/TEST/softwaremodules/294/artifacts/test-linux.raucb"    
              },
              "md5sum-http" : {
                "href" : "https://hawkbit.mydomain.se/DEFAULT/controller/v1/TEST/softwaremodules/294/artifacts/test-linux-delta.raucb.MD5SUM"
              }   
            }
          }
        ]
      }                                                       
    ]                                                         
  }                                                           
}                      
```

```

Bastian-Krause, 2022-02-01:
Okay, then this is expected behavior. rauc-hawkbit-updater queries the deploymentBase resource as advertised by your hawkBit instance. Quoting myself:

> the depoymentBase URL is used as given by hawkBit since 8624e71. Before that the URL was handcrafted. In my opinion the correct way is to use the URLs as provided by hawkBit.

[[1]](https://github.com/rauc/rauc-hawkbit-updater/issues/85#issuecomment-824645400)

 What makes hawkBit think that your instance is running on `hawkbit:8080`? If you fix your hawkBit configuration, rauc-hawkbit-updater will also work. I am not a hawkBit expert, but I assume that you can specify your FQDN and port somewhere.

ejoerns, 2022-02-01:
Maybe someone who successfully fixed this should add a hint to the documentation on how to fix/configure this in hawkBit.

einoj, 2022-02-01:
It was indeed the same issue as #85. 
This was the fix: `server.forward-headers-strategy=NATIVE` along with setting the correct incoming "Host:" header.

closed 2022-02-01 by einoj