rauc/rauc-hawkbit-updater #136: Handle network errors instead of starting rauc-hawkbit-updater after network-online.target

einoj, 2022-08-05:
This pull request https://github.com/rauc/rauc-hawkbit-updater/pull/86 sets rauc-hawkbit-updater to start after network-online.target. The PR links https://www.freedesktop.org/wiki/Software/systemd/NetworkTarget/ which states that waiting for network-online.target is a poor solution.

> $network is a mechanism that is required only to deal with software that assumes continuous network is available (i.e. of the simple non-well-written kind)

It goes on to state:

> If it cannot reach a server it must retry. If network configuration connectivity is lost it must react. 
...
>  A service written like this is robust, can be started at any time, and will always do the best of the circumstances it is running in.
...
> A robust system boots up independently of external services. More specifically, if a network DHCP server does not react this should not slow down boot on most setups, but only for those where network connectivity is strictly needed

IMO rauc-hawkbit-updater should handle the network errors and try again later rather than delaying startup like https://github.com/rauc/rauc-hawkbit-updater/pull/86 does. 

The current solution delays boot until:

> all configured network devices are up and have an IP address assigned

This has been problematic for us because we have a mobile modem that sometimes takes 2-3 minutes to connect, so even if we have a network connection on other interfaces (LAN/wifi), rauc-hawkbit-updater will delay the boot process waiting for the modem. Rauc-hawkbit-updater properly handling network errors instead of waiting for network-online.target would solve this issue.

Bastian-Krause, 2022-08-05:
Hi!

> This pull request #86 sets rauc-hawkbit-updater to start after network-online.target. The PR links https://www.freedesktop.org/wiki/Software/systemd/NetworkTarget/ which states that waiting for [...]

rauc-hawkbit-updater handles network errors just fine, in that case it tries again after a number of seconds specified in `retry_wait` (see [docs](https://rauc-hawkbit-updater.readthedocs.io/en/latest/reference.html#configuration-file)).

> The current solution delays boot until:
>
>
> This has been problematic for us because we have a mobile modem that sometimes takes 2-3 minutes to connect, so even if we have a network connection on [...]

Why is `After=network-online.target` wrong here? I agree that software should not rely on a working network connection, but I can't see any value added for rauc-hawkbit-updater to start earlier than at a time where least one interface is online. rauc-hawkbit-updater is an interface between the [RAUC D-Bus API](https://github.com/rauc/rauc) and the [hawkBit DDI API](https://github.com/eclipse/hawkbit), so there is really nothing to be done with no interface online. If you know your product might encounter late network interface configuration, you are free to add an individual systemd drop-in configuration to change that.

I see, however, that delayed boot might be an issue. We're happy to discuss this. Maybe @mschwan-phytec likes to join in?

What network daemon are you using? At least `systemd-networkd-wait-online.service` should be happy with one interface up and running. You should be able to configure this behavior in any case.

einoj, 2022-08-12:
Turns out this was just a misconfiguration by me. After enabling the correct "wait" service, rauc-hawkbit-updater starts as soon as any of the network interfaces are online. Great!

I also tested starting rauc-hawkbit-updater before the network was online whereupon it will print a failed to resolve message and try again later, so it works perfectly fine that way too.

closed 2022-08-12 by einoj

Bastian-Krause, 2022-08-12:
Thanks for the feedback, much appreciated!