rauc/rauc-hawkbit-updater #137: Improve reboot behaviour

BubuOT, 2022-08-30:
The [docs](https://rauc-hawkbit-updater.readthedocs.io/en/latest/reference.html) for the `post_update_reboot` flag currently state:

> Note that this results in an immediate reboot without contacting the system manager and without terminating any processes or unmounting any file systems. This may result in data loss.

This sounded a bit too scary for us, so we patched this to instead call `sudo systemctl reboot` via `system()` (with whitelisted `systemctl reboot` cmd in sudoers file). While that works, relying on sudo and calling the external isn't really great either. It seems that via dbus this is only possibly if you run [logind](https://www.freedesktop.org/wiki/Software/systemd/logind/) which we currently don't. In non-systemd systems running an external command (`/sbin/reboot`?) is probably the only solution anyway though.

I mainly wanted to start the discussion on finding an upstream solution to this problem here.

Can we automatically discover how the system should be rebooted (if logind is available, etc.)? Do we need a reboot-method config option?

ejoerns, 2022-08-30:
FTR: We had similar discussions in the PR (#39) that the option originates from:

https://github.com/rauc/rauc-hawkbit-updater/pull/39#issuecomment-766839767

One of the options I could think of is a D-Bus interface that allows us to notify another service/application which then handles the graceful shutdown.

BubuOT, 2022-11-03:
I've encountered another question/problem regarding reboots. We sometimes update our recovery partition via a separate update bundle (because it's updated a lot less frequently than the main system. Restarting after this bundle is installed makes of course no sense because we won't boot into the newly installed system.

I wonder how this could be solved though, any ideas?

Bastian-Krause, 2022-11-03:
> I've encountered another question/problem regarding reboots. We sometimes update our recovery partition via a separate update bundle (because it's updated a lot less frequently than the main system. [...]

I don't think `post_update_reboot=true` or any similar mechanism should be used in such a case.

I'd opt for some kind of application/daemon that can either be notified via a [post-install handler](https://rauc.readthedocs.io/en/latest/using.html#system-based-customization-handlers) or is subscribed to [D-Bus signals/properties](https://rauc.readthedocs.io/en/latest/reference.html#d-bus-api). A handler can [decide](https://rauc.readthedocs.io/en/latest/reference.html#custom-handlers-interface) whether a reboot is required. For the D-Bus case, the daemon/application can query [GetSlotStatus](https://rauc.readthedocs.io/en/latest/reference.html#gdbus-method-de-pengutronix-rauc-installer-getslotstatus) after the Completed signal and decide if a reboot is required. The application/daemon would trigger the reboot if required.

So I guess this is more a RAUC question than a rauc-hawkbit-updater question.

closed 2023-06-13 by BubuOT