rauc/rauc-hawkbit-updater #141: Compilation doesn't work since change to meson

lreinecke, 2022-10-18:
I am trying to compile this code, but everytime I run `meson setup build` ther is an error `ERROR: Neither directory contains a build file meson.build.`. I use this command in the base folder of the project `.../rauc-hawkbit-updater/`. Do you know how to fix this?

I use a crosscompiler. With cmake it worked fine. This schould not be a problem with meson?

Bastian-Krause, 2022-10-18:
Very strange. Could you please post the complete output of these commands?

```
$ rm -rf build
$ git status
$ git show -s
$ ls -la
$ meson setup --debug --wipe build
```

lreinecke, 2022-10-18:
I tried a bit with your commands. This was the result:
[image]
[image]
[image]
The `git show -s` resulted in this:
[image]

Bastian-Krause, 2022-10-18:
What version of meson are you using? Is there a wrapper around your meson or is it called directly?

You could add "strace" before meson to see where it's actually looking for the meson.build. But if these basic operations do not work as intended, I don't have high hopes that everything works as expected if this error is gone..

lreinecke, 2022-10-18:
It worked fine, without the crosscompiler. So its a problem with the crosscompiler, and not with your code.

Bastian-Krause, 2022-10-18:
Yes, I suspected that.

lreinecke, 2022-10-18:
Thanks for your time. I suppose I have to solve this problem fo my own.

Bastian-Krause, 2022-10-18:
If the solution might help others, please post it here. Thanks.

closed 2022-10-18 by Bastian-Krause

eli-schwartz, 2022-10-19:
> but everytime I run `meson setup build` ther is an error `ERROR: Neither directory contains a build file meson.build.`. I use this command in the base folder of the project `.../rauc-hawkbit-updater/`. Do you know how to fix this?

The two directories in question that "neither" contains a build file `meson.build` are either:
- the current working directory
- `build/`
or else, for versions of Meson before 0.42 (when `meson setup` was introduced):
- `setup/`
- `build/`

I believe it is vanishingly unlikely you are using versions of Meson older than 0.42, but at the same time you've strongly indicated that `ls` reports a meson.build file.

So it's probable that your cross compiler is actually overwriting the definition of the `meson` program, but posting the output of `type -a meson` (bash or zsh) or `whence -ca ls` (optional for zsh) could help clarify that.

lreinecke, 2022-10-20:
The meson of my crosscompiler is version 0.53.2
There is an `meson.build` file in the base directory of the project where it should be.
If I only type `meson setup` or `meson build` it finds the `meson.build` but throws the Error `ERROR: Unable to determine dynamic linker`
If I use `strace` it shows, that it looks only in the subdirectories `setup/` and `build/` for the file. There are none and so it fails.

lreinecke, 2022-10-21:
@eli-schwartz @Bastian-Krause I have found a solution for (all of) my problems. My crosscompiler overwrites the PATH variable in a way that it used a meson-binary with the version 0.53.2. This version didn't work for me. So i had to use the local meson by typing `/usr/bin/meson setup build --cross-file meson.cross`. Then it works fine.