rauc/rauc-hawkbit-updater #147: Allow overriding pollking time from config

BubuOT, 2022-10-24:
It would be helpful to have development/staging targets poll hawkbit more often than the server-side defined polling interval (which is set to multiple hours so we save mobile bandwidth).

Our development targets are connected via an ethernet connection though and it would ease the testing workflow if we could tell those to just poll hawkbit every minute instead.

ejoerns, 2022-10-24:
Since the config value provided by the DDI is only refered to as the '*suggested sleep time between polls*' it also sounds reasonable to me to override this wish if we know better.

Bastian-Krause, 2022-10-26:
If this is testing-specific, maybe this should be a command line option rather than a config option?

BubuOT, 2022-10-26:
> If this is testing-specific, maybe this should be a command line option rather than a config option?

That wouldn't work that well for us as the config file is device-specific (so can be different on test-devices) but the service invocation is defined in the firmware and identical on all our devices.

Bastian-Krause, 2022-11-03:
Hm, okay, I think we could also add that to the config.

I am not sure whether this should override the poll time announced by hawkBit or if we should use the lowest value?

Bastian-Krause, 2022-11-03:
Makes sense. PR welcome :)