rauc/rauc-hawkbit-updater #149: Fail to download artifact: g_file_test: assertion 'filename != NULL' failed

dcd-arnold, 2022-11-07:
Hi,

I am running the rauc-hawkbit-updater v1.3 together with hawkbit 0.3.0M7.
When I configure a rollout the `rauc-hawkbit-updater` crashes with:
```
g_file_test: assertion 'filename != NULL' failed
```

**Settings and Log:**
[rauc-hawkbit-updater.conf](https://github.com/rauc/rauc-hawkbit-updater/files/9950151/rauc-hawkbit-updater.conf.txt)
[fail-rauc-hawlbit-updater.log](https://github.com/rauc/rauc-hawkbit-updater/files/9950152/fail-rauc-hawlbit-updater.log.txt)

What am I doing wrong?

ejoerns, 2022-11-07:
Having a short look at this I bet you run into `process_deployment_cleanup()` being called unconditionally even if using streaming where `hawkbit_config->bundle_download_location` might not be set:
https://github.com/rauc/rauc-hawkbit-updater/blob/b772772f913892432838f7e4b4b6086628782d6c/src/hawkbit-client.c#L780

A workaround for you should be to set `bundle_download_location = /any/path` in your `rauc-hawkbit-updater.conf` first of all.

dcd-arnold, 2022-11-07:
Thanks for getting back to me that quickly. Adding `bundle_download_location = <something>` to the config helped. The documentation said *Option can be ommited if stream_bundle is enabled.*

PR #150 (2022-11-14): Fix Omitted Bundle Download Location For Streaming Installations
Since the integration of RAUC streaming support in rauc-hawkbit-updater [1] the `bundle_download_location` config option can be ommited when the `stream_bundle` config option is enabled [2].

The value of the `bundle_download_location` config option is still used in streaming installation code paths leading to errors such as:
```
g_file_test: assertion 'filename != NULL' failed
```
To fix this, exit early in bundle clean up code and omit the check for free space if the `stream_bundle` config option is enabled.

We should omit the [... cut]
closed 2022-11-14 by ejoerns