rauc/rauc-hawkbit-updater #151: rauc-hawkbit-updater with streaming can easily exceed hawkbit quota on action updates

zyga, 2022-11-23:
Hawkbit has a default quota, that each action can have up to 1K status updates.

With streaming enabled, this nearly instantly overflows the quota with a sufficiently large image.

> lis 23 13:18:42 odo hawkbit.hawkbit[963]: 2022-11-23 13:18:42.219  WARN 963 --- [tp1756726852-23] o.e.h.repository.jpa.utils.QuotaHelper   : Cannot assign 1 ActionStatus entities to Action '51' because of the configured quota limit 1000. Currently, there are 1001 ActionStatus entities assigned.

ejoerns, 2022-11-23:
Yes, this is known. We have documented here how to properly configure hawkbit for streaming:

https://rauc-hawkbit-updater.readthedocs.io/en/latest/reference.html (under `stream_bundle` option).

Probably its not the most visible place :smirk:

Bastian-Krause, 2022-11-23:
See also https://github.com/eclipse/hawkbit/issues/1249

zyga, 2022-11-23:
I just returned to comment on the issue that this is indeed much better documented in master. Sorry about the noise.

https://github.com/rauc/rauc-hawkbit-updater/commit/8728e989b740933c14f43dfcce66df75c393cc4f

zyga, 2022-11-23:
As an important bit of context, the problem is not fixable here, because it is hawkbit itself that is generating the extra status entries in response to range requests. The only way, at the moment, to avoid this is to disable the quota inside hawkbit (which is not great) or to somehow force hawkbit to provide a 3rd-party (e.g. CDN) URL for the artifact.

closed 2022-11-23 by zyga