rauc/rauc-hawkbit-updater #152: SSL peer certificate or SSH remote key was not OK

Sourabh-ALTEN, 2022-11-29:
Hello,

I have a `rauc-hawkbit-updater` client setup to download software artifacts from HawkBit server. But, I am getting below error from `rauc-hawkbit-updater` about SSL verification. The same client is also present on other devices and communicating to same HawkBit server. And `rauc-hawkbit-updater` is working fine on them.
```
rauc-hawkbit-updater[6797]: Checking for new software...
rauc-hawkbit-updater[6797]: HTTP Error: HTTP request failed: SSL peer certificate or SSH remote key was not OK
```

The client is part of `systemd` service and is called like 
`/usr/bin/rauc-hawkbit-updater -s -c /etc/hawkbit/config.cfg`

I tried to get the `debug` output with `-d` flag. But, the `-d` option is not outputting anything (client is not executed). And I see that `verbose` mode is not enabled. I think, it would be better to include this to get more debug messages. 
```
/* ask libcurl to show us the verbose output */
  curl_easy_setopt(curl, CURLOPT_VERBOSE, 1L);
```
Should this be inluded in [hawkbit-client.c](https://github.com/rauc/rauc-hawkbit-updater/blob/767a8f8c4dcea8a31385de34d9c03871eaea25cf/src/hawkbit-client.c#L258)?

Can anyone please let me know what could be the issue here?

Thank you,
Sourabh

Bastian-Krause, 2022-11-29:
Hi,

> I have a `rauc-hawkbit-updater` client setup to download software artifacts from HawkBit server. But, I am getting below error from `rauc-hawkbit-updater` about SSL verification. The same client is [...]

This does not sound right. Where are you putting the `-d`? If you are modifying the systemd service, you need to `systemctl daemon-reload` and then restart your service. Anyway, this won't help you with your issue.

> And I see that `verbose` mode is not enabled. I think, it would be better to include this to get more debug messages.
>
> ```
> /* ask libcurl to show us the verbose output */
>   curl_easy_setopt(curl, [...]

https://curl.se/libcurl/c/CURLOPT_VERBOSE.html states that "you hardly ever want this set in production use, you will almost always want this when you debug/report problems." I consider code run with `-d` still production.

> Can anyone please let me know what could be the issue here?

The error message already tells you what's wrong. This is no rauc-hawkbit-updater issue. If you want to debug this further, use curl. You can pass the authentication header with `--header`, see https://www.eclipse.org/hawkbit/concepts/authentication/.

Sourabh-ALTEN, 2022-11-30:
Hi @Bastian-Krause thank you for the update.
Yes, indeed it was related to Hawkbit server certificate. Issue is resolved now.

closed 2022-11-30 by Sourabh-ALTEN