rauc/rauc-hawkbit-updater #180: Partial download test cases fail

tduf, 2024-06-28:
Despite my PR #179, there is still 2 test cases in failure when I run locally the `test_download.py`:
 * `test/test_download.py::test_download_partials_without_resume`: expected to fail but the download succeeds.
 * `test/test_download.py::test_download_partials_with_resume`: expected a download resuming but succeeds in one shot.

```
./test/wait-for-hawkbit-online && dbus-run-session -- pytest -v test/test_download.py 
Waiting for hawkbit to come up 
============================================================================================================ test session starts =============================================================================================================
platform linux -- Python 3.12.3, pytest-8.2.2, pluggy-1.5.0 -- /home/thibaud/Work/xxx/repo/rauc-hawkbit-updater/venv/bin/python3
cachedir: .pytest_cache
rootdir: /home/thibaud/Work/xxx/repo/rauc-hawkbit-updater
configfile: pytest.ini
collected 7 items                                                                                                                                                                                                                            

test/test_download.py::test_download_inexistent_location PASSED                                                                                                                                                                        [ 14%]
test/test_download.py::test_download_unallowed_location PASSED                                                                                                                                                                         [ 28%]
test/test_download.py::test_download_too_slow PASSED                                                                                                                                                                                   [ 42%]
test/test_download.py::test_download_partials_without_resume FAILED                                                                                                                                                                    [ 57%]
test/test_download.py::test_download_partials_with_resume FAILED                                                                                                                                                                       [ 71%]
test/test_download.py::test_download_slow_with_resume PASSED                                                                                                                                                                           [ 85%]
test/test_download.py::test_download_only PASSED              
```

Full logs: 
[rauc-hawkbit-updater-download-issue-full-logs.txt](https://github.com/user-attachments/files/16032457/rauc-hawkbit-updater-download-issue-full-logs.txt)

Bastian-Krause, 2024-12-18:
Strange, could you please pass `-s` to pytest and provide the logs? Maybe the nginx log level needs to be adjusted to get useful info about this.

tduf, 2024-12-19:
Here is a run on the PR #179 branch with Fedora 41 (nginx 1.26.2) with pytest -s: 
[failure_with_-s_on_pr179.log](https://github.com/user-attachments/files/18195245/failure_with_-s_on_pr179.log)

Bastian-Krause, 2025-03-12:
@tduf Could you try #188?

PR #188 (2025-07-10): test: rework nginx partial download fixture, move nginx config snippets to dedicated files
Until now, the partial download tests did not work reliably and relied on implicit nginx behavior: `limit_rate_after 200k;` combined with `limit_rate 70k;` lead to nginx sending "206 Partial Content" responses. As far as I remember, this was found by trial and error.

Let's use a more stable solution: lua scripting allows us to control reliable and fine grained what hawkBit artifacts nginx serves. So add a config snippet that makes nginx only serve the first half of the RAUC bundle, then close the connec [... cut]